cambridgehackers/connectal #54: DMA burst size limited to 32

hanw, 2015-02-13:
In example tests/memread_manual, variable burstLen has no effect on the length of DMA transaction burst len. 

I added the follow lines to Rtest.bsv
+  61    rule cyc;
+  62       cycle <= cycle + 1;                                                                                                              
+  63    endrule
   70    rule check;                                                                                                                         
  71       let v <- toGet(re.dataPipes[0]).get;
  72       let expectedV = {srcGens+1,srcGens};                                                                                             
  73       let misMatch = v != expectedV;                                                                                                   
  74       mismatchCounts <= mismatchCounts + (misMatch ? 1 : 0);                                                                           
  75       let new_srcGens = srcGens+2;
  76       if (new_srcGens >= truncate(chunk/4))
  77 >....... new_srcGens = 0;                                                                                                              
  78       srcGens <= new_srcGens;                                                                                                          
+  79       $display("%d: check data %h", cycle, v);
  80    endrule 

Example Log: 
    67940: check data 0000000100000000
    67941: check data 0000000300000002
    67942: check data 0000000500000004
    67943: check data 0000000700000006
    67944: check data 0000000900000008
    67945: check data 0000000b0000000a
    67946: check data 0000000d0000000c
    67947: check data 0000000f0000000e
    67948: check data 0000001100000010
    67949: check data 0000001300000012
    67950: check data 0000001500000014
    67951: check data 0000001700000016
    67952: check data 0000001900000018
    67953: check data 0000001b0000001a
    67954: check data 0000001d0000001c
    67955: check data 0000001f0000001e
    67956: check data 0000002100000020
    67957: check data 0000002300000022
    67958: check data 0000002500000024
    67959: check data 0000002700000026
    67960: check data 0000002900000028
    67961: check data 0000002b0000002a
    67962: check data 0000002d0000002c
    67963: check data 0000002f0000002e
    67964: check data 0000003100000030
    67965: check data 0000003300000032
    67966: check data 0000003500000034
    67967: check data 0000003700000036
    67968: check data 0000003900000038
    67969: check data 0000003b0000003a
    67970: check data 0000003d0000003c
    67971: check data 0000003f0000003e
    68126: check data 0000004100000040
    68127: check data 0000004300000042
    68128: check data 0000004500000044
    68129: check data 0000004700000046
    68130: check data 0000004900000048
    68131: check data 0000004b0000004a

jameyhicks, 2015-02-13:
I had to increase memread buffer size to hold a larger burst. So I made the default size a function of BurstLenSize.

I think this is fixed now. Could you test?

hanw, 2015-02-13:
Could you take a look at the debug_memread branch that I push to my clone?

jameyhicks, 2015-02-18:
I'm adding a makefile variable so that BurstLenSize can be changed without editing the code.

hanw, 2015-02-18:
Thanks! I need longer BurstLenSize to simulate/debug the PHY layer design, as the PHY layer needs a continuous stream of data on every clock tick. However the BurstLenSize is currently limited to 10 in MemSlaveEngine by TLPLength, which means I can only simulate up to 128(?) clock cycles. I guess this is sufficient for small unit-test cases. But it would be nice to be able to bypass the limit of 10 to enable longer simulations.

jameyhicks, 2015-02-18:
TLPLength is a PCIe parameter

jameyhicks, 2015-02-18:
I'm going to mark this closed. We'll have to look at PCIe to see how to achieve the throughput you need.

closed 2015-02-18 by jameyhicks