cambridgehackers/connectal #55: generate xilinx pin constraints from .pkg file

jameyhicks, 2015-02-13:
e.g.,

Xilinx/Vivado/2014.4/data/parts/xilinx/zynq/zynq/xc7z020/clg400/clg400_7z020.pkg:

N2           | 148        PS_DDR_A0_502                 N2
K2           | 149        PS_DDR_A1_502                 K2
M3           | 150        PS_DDR_A2_502                 M3
K3           | 151        PS_DDR_A3_502                 K3

jameyhicks, 2015-02-13:
Larry found a reference to partgen, but that was part of the old tool suite. The Vivado migration guide points to some tcl commands that look interesting:
   http://www.xilinx.com/support/documentation/sw_manuals/xilinx2012_3/ug911-vivado-migration.pdf

jankcorn, 2015-02-13:
How about using the file:
Xilinx/Vivado/2014.4/data/ip/xilinx/mig_7series_v2_3/data/fpga_tlib/zynq/7z/xc7z020clg400_pkg.xml
?
It looks much easier to parse (just standard XML)
I assume we are just trying to get the memory pins and this is part of the MemoryInterfaceGenerator support files.

jameyhicks, 2015-02-13:
It's true the format is very goofy.

I think if we can write a short tcl script that uses Xilinx's parser for their package files it would be better.

jankcorn, 2015-02-17:
deleted the package xdc files and all builds seem ok.
Turns out weren't needed after all.

lstewart, 2015-02-18:
I now think they are needed.  Looking at, for example, examples/simple/zedboard/Impl/TopDown/pre_place_drc.rpt we see

NSTD-1#1 Critical Warning
Unspecified I/O Standard
130 out of 138 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may
cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extre
me cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O stand
ards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. T
o allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warn
ing} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command
to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: DDR_Addr[14:0
], DDR_BankAddr[2:0], DDR_DM[3:0], DDR_DQ[31:0], DDR_DQS_n[3:0], DDR_DQS_p[3:0], FIXED_IO_mio[53:0], DDR_CAS_n, DDR_CKE, DDR_CS_
n, DDR_Clk_n, DDR_Clk_p, DDR_DRSTB, DDR_ODT, DDR_RAS_n (the first 15 of 21 listed).
Related violations: <none>

This probably has been set as a warning but it should be an error.
I am back looking for a way to generate the necessary xdc files