cambridgehackers/connectal #115: Issues with defined macros in bsvpreprocessor.py

acw1251, 2016-05-09:
Right now, the following is not supported in connectal/scripts/bsvpreprocess.py (code from lib/BSVSource/Contexts/Contexts.defines within the Bluespec installation directory):
```
`define SynthBoundary(mkM,IM) \
(*synthesize*)\
module [Module] mkM``V(Tuple2#(completeContextIfc,IM))\
  provisos(Expose#(CompleteContext,completeContextIfc,_n));\
  (*hide*)\
  let _init<-mkInitialCompleteContext;\
  (*hide*)\
  let _ifc <- unbury(_init,mkM``0);\
  return _ifc;\
endmodule\
module [ModuleContext#(CompleteContext)] mkM(IM);\
  (*hide*)\
  let _ifc<-rebury(mkM``V);\
  return _ifc;\
endmodule
```
The current error produced by bsvpreprocessor.py is that there is an unhandled preprocessor token on line 3 (but that `` is just for concatenation after substitution).

I took a look at the code, and it looks like it doesn't support macros with arguments. In addition to that, it looks like it is only taking the next token after the macro as its definition, but I believe that it should take the rest of the line as the definition.

jankcorn, 2016-05-09:
I would highly recommend that you not pass this file in 'BSVFILES=xxx',
since the macro is not directly used in an interface definition (which is
all the preprocessor is trying to handle).

Instead, add '--bsvpath' to CONNECTALFLAGS so that bsc can locate the file
from an 'import' statement.

Sorry, but the preprocessor handling is pretty primitive...

acw1251, 2016-05-09:
It wasn't in BSVFILES. Actually the file causing the problem was just some file in --bsvpath that wasn't being used anywhere so I deleted it. I do have some other code that looks similar, and I may run into similar problems with it later, but for now I'm just getting c++ compile errors (tons of them because I'm in the middle of refactoring all of our c++ code).