cambridgehackers/connectal #124: build error with netfpga sume

tudang, 2016-08-15:
Hi,

I'm trying to build the DDR3 test for nfsume target. The build was failed with the following error. Maybe you forget to check-in some file.
```
Error: XCI file /opt/connectal/out/nfsume/axiddr3/axiddr3.xci not found
make[1]: *** [fpgamake.mk] Error 255
make: *** [build.nfsume] Error 2
```

jameyhicks, 2016-08-15:
The Makefile includes a rule for generating the IP core, but it is not run by default. We can fix that, but for now:

make ip.nfsume ; make build.nfsume

tudang, 2016-08-16:
Thank Jamey. It solved my issue.

jameyhicks, 2016-08-16:
@tudang please keep me posted and let me know if there's anything I can do to help.

tudang, 2016-08-16:
Hi Jamey,

after generating the IP for nfsume and building, I got the following error:

```
create_ip -name pcie3_7x -version 4.2 -vendor xilinx.com -library ip -module_name pcie3_7x_0 -dir /opt/connectal/out/nfsume
ERROR: [Coretcl 2-1133] Requested IP 'xilinx.com:ip:pcie3_7x:4.2' cannot be created. The latest available version in the catalog is 'xi
linx.com:ip:pcie3_7x:4.1'. If you do not wish to select a specific version please omit the version field from the command arguments, or
 use a wildcard in the VLNV.
INFO: [Common 17-206] Exiting Vivado at Tue Aug 16 13:56:02 2016...
make[1]: *** [prebuild] Error 1
make: *** [gen.nfsume] Error 2
```

If I modify the script `connectal-synth-pcie.tcl` to use pcie3 version 4.1, then it works. Do I need to generate something else for solving this error?

jameyhicks, 2016-08-16:
That script has a check for the version of vivado and chooses the PCIe
version based on that. I guess we need to update it for the version of
vivado you are using.

I'm currently using vivado 2016.2 but I have also tested with vivado 2016.1.

I think you can also specify version 4.* which might be easier.

tudang, 2016-08-16:
For your information, I'm using vivado 2015.4.

jameyhicks, 2016-08-16:
Looks like I had a copy paste error when updating to 2016.x and have not tested the gen3 core since then.

I just pushed a change which should fix this. I should have included the issue number in the commit.

jameyhicks, 2016-08-16:
That was commit d231954ab394eddc8a9a1ebdbd78dc3f22050b42

closed 2016-08-19 by tudang