cambridgehackers/connectal #144: pcieportal.ko issue

hanw, 2017-04-20:
When compiling connectal kernel driver on redhat enterprise linux, the following error happens.

```
(sonic_proj) [jschuh@worthy-one pcieportal]$ make
md5sum *.c ../../generated/cpp/*.c | grep -v mod.c | sed -f ../../scripts/driver_signature.sed >pcieportal_signature_file.h
make[1]: Entering directory `/usr/src/kernels/3.10.105-<tel:3.10.105-1>1<tel:3.10.105-1>.el6.elrepo.x86_64'
 CC [M]  /home/jschuh/work/cs6410-project/connectal/drivers/pcieportal/pcieportal.o
 Building modules, stage 2.
 MODPOST 1 modules
WARNING: "pcie_get_mps" [/home/jschuh/work/cs6410-project/connectal/drivers/pcieportal/pcieportal.ko] undefined!
WARNING: "pcie_set_mps" [/home/jschuh/work/cs6410-project/connectal/drivers/pcieportal/pcieportal.ko] undefined!
 LD [M]  /home/jschuh/work/cs6410-project/connectal/drivers/pcieportal/pcieportal.ko
make[1]: Leaving directory `/usr/src/kernels/3.10.105-1.el6.elrepo.x86_64'
```

jameyhicks, 2017-04-20:
In that version of the kernel, pcie_get_mps and pcie_set_mps are not exported for use in modules.

Commit 49c4ebd14f4e6785a10af39b43c078f3bfcfc6a4 solves this problem by conditionally compiling the code that tunes the PCIE capabilities only if PCIEPORTAL_TUNE_CAPS is defined.

jameyhicks, 2017-04-20:
Connectal advertises a larger payload size in its PCIe configuration registers, so in theory the driver should not have to do this, which is why the symbols were not exported.

However, BIOSes being what they are, sometimes the driver does have to give the system a nudge, which is why I introduced this code. It does enable the FPGA to achieve higher throughput for DMA when supported.

That being said, if you need higher performance from the FPGA and the system's chipset supports larger payload sizes than configured by the BIOS, I would recommend a newer kernel.

closed 2020-04-22 by jameyhicks