cambridgehackers/connectal #145: pcieportal.c compile issue with Linux kernel 4.10

demetriussuchkov, 2017-09-11:
I get the following error when compiling connectal on Ubuntu 16.04 LTS (kernel 4.10.0-28-generic (x86_64)):
```
dmitry@ubuntu:~/projects/connectal$ make all && sudo make install
(cd drivers/pcieportal; make)
make[1]: Entering directory '/home/dmitry/projects/connectal/drivers/pcieportal'
md5sum *.c ../../generated/cpp/*.c | grep -v mod.c | sed -f ../../scripts/driver_signature.sed >pcieportal_signature_file.h
make[2]: Entering directory '/usr/src/linux-headers-4.10.0-28-generic'
  CC [M]  /home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.o
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:725:8: error: type defaults to ‘int’ in declaration of ‘DEFINE_PCI_DEVICE_TABLE’ [-Werror=implicit-int]
 static DEFINE_PCI_DEVICE_TABLE(pcieportal_id_table) = {{
        ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:725:1: warning: parameter names (without types) in function declaration
 static DEFINE_PCI_DEVICE_TABLE(pcieportal_id_table) = {{
 ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:725:1: error: function ‘DEFINE_PCI_DEVICE_TABLE’ is initialized like a variable
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:725:1: warning: braces around scalar initializer
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:725:1: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
In file included from /home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:28:0:
./include/linux/pci.h:725:2: error: field name not in record or union initializer
  .vendor = (vend), .device = (dev), \
  ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
         PCI_DEVICE(BLUESPEC_VENDOR_ID, CONNECTAL_DEVICE_ID)}, { /* end: all zer
         ^
./include/linux/pci.h:725:2: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
  .vendor = (vend), .device = (dev), \
  ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
         PCI_DEVICE(BLUESPEC_VENDOR_ID, CONNECTAL_DEVICE_ID)}, { /* end: all zer
         ^
./include/linux/pci.h:725:12: error: invalid initializer
  .vendor = (vend), .device = (dev), \
            ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
         PCI_DEVICE(BLUESPEC_VENDOR_ID, CONNECTAL_DEVICE_ID)}, { /* end: all zer
         ^
./include/linux/pci.h:725:12: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
  .vendor = (vend), .device = (dev), \
            ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
         PCI_DEVICE(BLUESPEC_VENDOR_ID, CONNECTAL_DEVICE_ID)}, { /* end: all zer
         ^
./include/linux/pci.h:725:20: error: field name not in record or union initializer
  .vendor = (vend), .device = (dev), \
                    ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
         PCI_DEVICE(BLUESPEC_VENDOR_ID, CONNECTAL_DEVICE_ID)}, { /* end: all zer
         ^
./include/linux/pci.h:725:20: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
  .vendor = (vend), .device = (dev), \
                    ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
         PCI_DEVICE(BLUESPEC_VENDOR_ID, CONNECTAL_DEVICE_ID)}, { /* end: all zer
         ^
./include/linux/pci.h:725:30: warning: excess elements in scalar initializer
  .vendor = (vend), .device = (dev), \
                              ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
         PCI_DEVICE(BLUESPEC_VENDOR_ID, CONNECTAL_DEVICE_ID)}, { /* end: all zer
         ^
./include/linux/pci.h:725:30: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
  .vendor = (vend), .device = (dev), \
                              ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
         PCI_DEVICE(BLUESPEC_VENDOR_ID, CONNECTAL_DEVICE_ID)}, { /* end: all zer
         ^
./include/linux/pci.h:726:2: error: field name not in record or union initializer
  .subvendor = PCI_ANY_ID, .subdevice = PCI_ANY_ID
  ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:726:9: note: in expansion of macro ‘PCI_DEVICE’
[... 114 lines cut ...]
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:849:16: note: in expansion of macro ‘DRIVER_VERSION’
 MODULE_VERSION(DRIVER_VERSION);
                ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:725:8: warning: ‘DEFINE_PCI_DEVICE_TABLE’ defined but not used [-Wunused-function]
 static DEFINE_PCI_DEVICE_TABLE(pcieportal_id_table) = {{
        ^
In file included from /home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:26:0:
./include/linux/module.h:213:27: error: ‘__mod_pci__pcieportal_id_table_device_table’ aliased to undefined symbol ‘pcieportal_id_table’
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                           ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:728:1: note: in expansion of macro ‘MODULE_DEVICE_TABLE’
 MODULE_DEVICE_TABLE(pci, pcieportal_id_table);
 ^
cc1: some warnings being treated as errors
scripts/Makefile.build:301: recipe for target '/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.o' failed
make[3]: *** [/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.o] Error 1
Makefile:1524: recipe for target '_module_/home/dmitry/projects/connectal/drivers/pcieportal' failed
make[2]: *** [_module_/home/dmitry/projects/connectal/drivers/pcieportal] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.10.0-28-generic'
Makefile:38: recipe for target 'pcieportal.ko' failed
make[1]: *** [pcieportal.ko] Error 2
make[1]: Leaving directory '/home/dmitry/projects/connectal/drivers/pcieportal'
Makefile:32: recipe for target 'pciedrivers' failed
make: *** [pciedrivers] Error 2
```

jankcorn, 2017-09-11:
Hi,

I pushed a 'possible' fix for this (unfortunately, I don't have a machine
with 16.04LTS installed).

Could you please pull/try again?
Thanks!
jca

demetriussuchkov, 2017-09-12:
Hello!
Now I get the following error:
```
dmitry@ubuntu:~/projects/connectal$ make all
(cd drivers/pcieportal; make)
make[1]: Entering directory '/home/dmitry/projects/connectal/drivers/pcieportal'
md5sum *.c ../../generated/cpp/*.c | grep -v mod.c | sed -f ../../scripts/driver_signature.sed >pcieportal_signature_file.h
make[2]: Entering directory '/usr/src/linux-headers-4.10.0-33-generic'
  CC [M]  /home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.o
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c: In function ‘pcieportal_init’:
<command-line>:0:16: error: implicit declaration of function ‘KBUILD_STR’ [-Werror=implicit-function-declaration]
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:814:87: note: in expansion of macro ‘DRIVER_VERSION’
 N_INFO "%s: Registered Connectal Pcieportal driver %s\n", DEV_NAME, DRIVER_VERS
                                                                     ^
<command-line>:0:27: error: too many decimal points in number
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:814:87: note: in expansion of macro ‘DRIVER_VERSION’
 N_INFO "%s: Registered Connectal Pcieportal driver %s\n", DEV_NAME, DRIVER_VERS
                                                                     ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c: In function ‘pcieportal_exit’:
<command-line>:0:27: error: too many decimal points in number
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:838:89: note: in expansion of macro ‘DRIVER_VERSION’
 INFO "%s: Unregistered Connectal Pcieportal driver %s\n", DEV_NAME, DRIVER_VERS
                                                                     ^
In file included from ./include/linux/module.h:18:0,
                 from /home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:26:
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c: At top level:
<command-line>:0:16: error: expected ‘,’ or ‘;’ before ‘KBUILD_STR’
./include/linux/moduleparam.h:23:26: note: in definition of macro ‘__MODULE_INFO’
   = __stringify(tag) "=" info
                          ^
./include/linux/module.h:237:34: note: in expansion of macro ‘MODULE_INFO’
 #define MODULE_VERSION(_version) MODULE_INFO(version, _version)
                                  ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:854:1: note: in expansion of macro ‘MODULE_VERSION’
 MODULE_VERSION(DRIVER_VERSION);
 ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:854:16: note: in expansion of macro ‘DRIVER_VERSION’
 MODULE_VERSION(DRIVER_VERSION);
                ^
<command-line>:0:27: error: too many decimal points in number
./include/linux/moduleparam.h:23:26: note: in definition of macro ‘__MODULE_INFO’
   = __stringify(tag) "=" info
                          ^
./include/linux/module.h:237:34: note: in expansion of macro ‘MODULE_INFO’
 #define MODULE_VERSION(_version) MODULE_INFO(version, _version)
                                  ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:854:1: note: in expansion of macro ‘MODULE_VERSION’
 MODULE_VERSION(DRIVER_VERSION);
 ^
/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.c:854:16: note: in expansion of macro ‘DRIVER_VERSION’
 MODULE_VERSION(DRIVER_VERSION);
                ^
cc1: some warnings being treated as errors
scripts/Makefile.build:301: recipe for target '/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.o' failed
make[3]: *** [/home/dmitry/projects/connectal/drivers/pcieportal/pcieportal.o] Error 1
Makefile:1524: recipe for target '_module_/home/dmitry/projects/connectal/drivers/pcieportal' failed
make[2]: *** [_module_/home/dmitry/projects/connectal/drivers/pcieportal] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.10.0-33-generic'
Makefile:38: recipe for target 'pcieportal.ko' failed
make[1]: *** [pcieportal.ko] Error 2
make[1]: Leaving directory '/home/dmitry/projects/connectal/drivers/pcieportal'
Makefile:32: recipe for target 'pciedrivers' failed
make: *** [pciedrivers] Error 2
```

jameyhicks, 2017-09-12:
I installed the 4.10 hwe kernel so I can debug this.

jameyhicks, 2017-09-12:
Commit 0444c3d36530be3f8096de8bfe0d20b09e060bec fixes the build. I have not had a chance to test it yet.

jameyhicks, 2017-09-12:
I packaged this as version 17.09.1 -- launchpad is building packages now.

closed 2020-04-22 by jameyhicks