cambridgehackers/connectal #156: Adding Connectal Support to DE5 Net Board

NP95, 2020-01-27:
I have started using Bluespec and I am really awed by the change in paradigm I am having writing HDL. So I have a project in which I plan to use Bluespec, however the board I have is an Altera DE5 Net board, if I wanted to add support to the board what are the steps I have to do?

hanw, 2020-01-27:
DE5 board is 'unofficially' supported. It is not mentioned in the README.

The board type is 'de5', for example, to build the echo example, you can use:
`make -C examples/echo build.de5`

You can try to search for the PRs made by hanw back in 2014 - 2015, that's when the support for DE5 was added.

jameyhicks, 2020-04-22:
@NP95 did you have any luck with the DE5 board?

NP95, 2020-04-22:
Hi Jamey,

Immediately after that I was busy in other work and after that I haven't
had access to the board in the last 2.5 months. Will update once I have
access to it.

hanw, 2020-04-22:
@NP95 Note that last time DE5 was tested was using Quartus 15.1. I have just tried Quartus 19.2 a few days ago and found that the qsys-generate tool has changed quite a bit and the existing tcl script for for de5 is no longer compatible. You will need to spend a bit of time bring the tcl script up-to-date.