cambridgehackers/connectal #157: Connectal Support for open source bsc compiler

rajat-agarwal-, 2020-02-20:
I tried using connectal with [open-source version of Bluespec](https://github.com/B-Lang-org/bsc/) and it fails.
There seems to be some outdated switches in connectal and include locations in regards to open sourced version.
Attached are the error logs.

Kindly help.
[log_V.txt](https://github.com/cambridgehackers/connectal/files/4232365/log_V.txt)
[log.txt](https://github.com/cambridgehackers/connectal/files/4232367/log.txt)

jameyhicks, 2020-02-20:
I have not been able to build it so I have not tested it yet. I will pull from the repo and try again.

Thank you for sharing the log files, these are helpful.

hanw, 2020-02-21:
I tried to build echo for bluesim. The opensource version seems to miss some Xilinx libraries in Prelude. For instance, the XilinxCells.bo file cannot be found.

hanw, 2020-02-22:
This is resolved with the release of contrib library

rajat-agarwal-, 2020-02-23:
I am still getting the following error if i try to build echo for bluesim.
make[1]: *** No rule to make target 'obj/Vector.bo', needed by 'obj/XsimTop.bo'.  Stop.

hanw, 2020-02-23:
You can use `V=1 make build.bluesim` to print the commands used by connectal to compile your module. Current problem is the licensed bsc release and open-source bsc release install the library file to different paths. In the licensed bsc, all library files are installed under `Prelude`, in open-source bsc, the library files are under `Libraries` and the contrib library files are under their own directory. We need to update connectal to source from all these places.

hanw, 2020-02-23:
Search for `Prelude` in connectal repo, these two python files need fix. `scripts/packagesource.py`, `scripts/bsvdependencies.py`  Can you perhaps make a PR?

jameyhicks, 2020-03-03:
bsc-contrib has XilinxCells but the hierarchy has changed, so one more fix needed.

jameyhicks, 2020-03-03:
I was able to build and run examples/simple for verilator using open source bsc and bsc-contrib.
I compiled the BSV to Verilog for VC707 (build.vc707g2) and I'm compiling the verilog to a bitstream now.

This depends on bsc and bsc-contrib being installed into the same prefix:
`cd bsc; make PREFIX=~/Bluespec install`
`cd bsc-contrib; make PREFIX=~/Bluespec install`
`export BLUESPECDIR=~/Bluespec/lib`
`cd connectal/examples/simple; make build.verilator run.verilator`

We can make another change so that if `bsc` is on the path but `BLUESPECDIR` is not that the scripts would detect where the libraries are by running `which bsc`.

closed 2020-03-03 by jameyhicks

acw1251, 2020-03-03:
Just using `which bsc` will not work if bsc is a sym link. Instead you can use the technique mentioned in https://github.com/B-Lang-org/bsc/pull/33 to get the library folder for open-sourced bsc. Running the command `echo 'puts $env(BLUESPECDIR)' | bluetcl` in the shell will get you the internal value of BLUESPECDIR used in bluetcl. This command works with licensed bsc as well.

One more thing, since bsc-contrib is separate from bsc now, would it make sense to add bsc-contrib as a submodule to connectal?

jameyhicks, 2020-03-03:
I was planning to dereference any symlinks. But why use a symlink rather than adjusting PATH?

I'm working on making bsc and bsc-contrib .debs.

commit 674fee64 (2020-03-03): Open source bsc libraries have more hierarchy

So search several levels down for dependences and add Libraries/FPGA/* to the bsc search path.

Commit [3/3] that fixes #157.