cambridgehackers/connectal #169: Is it possible to support the opensource bsc?

kurocatus, 2020-06-05:
I am trying to use the [opensourced version](https://github.com/B-Lang-org/bsc) of bluespec compiler together with connectal. However, in order to successfully build the project (generating bluesim executables), I have to make some changes to connectal (see attached file).

Building errors are due to directory organization and internal API difference between closesource and opensourced version bsc. So I would like to know if there is any plan for connectal to support the opensourced version, since it seems to be the most accessible one for most people and hopefully Bluespec Inc. will continually maintain the repo.
[connectal.txt](https://github.com/cambridgehackers/connectal/files/4734136/connectal.txt)

jameyhicks, 2020-06-05:
I've been using the opensource bluespec compiler with Connectal for a few months now. However, I have only tested verilator simulation.

Thank you for providing the change to make bluesim work with the opensource version. Would you like to make a pull request?

kurocatus, 2020-06-06:
Applying the patch without further modifications would actually make connectal break if someone is using an older bluespec compiler (versions in README.md), so I don't know if it is a good idea to make a pull request right away.

It would be better to get the bsc version and choose different codes depending on that. But how should I get bsc version reliably? `bsc -help` doesn't give any useful information.

jameyhicks, 2020-06-07:
I don't know if I have any users that depend on closed source bsc so I
don't know how much work should be put into supporting it.

Also, the open source bsc doesn't have version numbers yet.

jankcorn, 2020-06-08:
I  also think that there is not much value in continuing support for
the closed source bsc.
Despite this, I think there is definitely a long term need for the
open source bsc to have compiler version numbers.

I created a pull request for cambridgehackers/bsc to give an idea of
the changes required.
I am not sure the best way to actually manage/update the environment
variable "COMPILERVERSION" (I have just dropped the definition into
src/comp/Makefile temporarily to show how it _could_ work)

BTW, if we have a compiler version string, then putting in conditional
compilation for the bluesim changes should be trivial.

Ideas?
jca

kurocatus, 2020-06-10:
A tentative pull request #170

closed 2020-06-11 by kurocatus