cambridgehackers/connectal #184: PCIe drivers fail to compile on Ubuntu 16.04

CyanoKobalamyne, 2021-07-15:
When trying to compile `pcieportal`, I get the following error:
```
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c: In function ‘pcieportal_ioctl’:
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c:204:119: error: missing binary operator before token "("
 #if (LINUX_VERSION_CODE < KERNEL_VERSION(5,0,0)) && !(defined(RHEL_MAJOR) && RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0))
                                                                                                                       ^
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c:207:69: error: macro "access_ok" requires 3 arguments, but only 2 given
                 err = !access_ok((void __user *) arg, _IOC_SIZE(cmd));
                                                                     ^
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c:207:24: error: ‘access_ok’ undeclared (first use in this function)
                 err = !access_ok((void __user *) arg, _IOC_SIZE(cmd));
                        ^
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c:207:24: note: each undeclared identifier is reported only once for each function it appears in
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c:210:119: error: missing binary operator before token "("
 #if (LINUX_VERSION_CODE < KERNEL_VERSION(5,0,0)) && !(defined(RHEL_MAJOR) && RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0))
                                                                                                                       ^
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c:213:69: error: macro "access_ok" requires 3 arguments, but only 2 given
                 err = !access_ok((void __user *) arg, _IOC_SIZE(cmd));
                                                                     ^
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c: In function ‘pcieportal_board_activate’:
/home/ubuntu/src/connectal/drivers/pcieportal/portal.c:548:39: error: implicit declaration of function ‘pci_irq_vector’ [-Werror=implicit-function-declaration]
                 this_board->irq_num = pci_irq_vector(this_board->pci_dev, 8);
                                       ^
```
Kernel version is the default 4.4.0. I also tried on 18.04, with a newer kernel (the default, I think 4.15), and that gave me the same error.

CyanoKobalamyne, 2021-07-15:
The version from the repo seems to work on 18.04, but on 16.04 that also fails to install, although with a different error:
```
/var/lib/dkms/connectal/20.06.4/build/portal.c: In function ‘pcieportal_board_activate’:
/var/lib/dkms/connectal/20.06.4/build/portal.c:548:39: error: implicit declaration of function ‘pci_irq_vector’ [-Werror=implicit-function-declaration]
                 this_board->irq_num = pci_irq_vector(this_board->pci_dev, 8);
                                       ^
```

manish-pp, 2022-08-20:
Do you have any idea how to resolve this error in ubuntu 16.04?

> The version from the repo seems to work on 18.04, but on 16.04 that also fails to install, although with a different error:
>
> ```
> /var/lib/dkms/connectal/20.06.4/build/portal.c: In function [...]

jameyhicks, 2022-08-22:
Hi Manish,

Ubuntu 16.04 is no longer getting security updates, so I recommend upgrading to the newest LTS that you can use. 

Otherwise, the thing to do is look up pci_irq_vector in Linux cross reference and look at its history between 16.04 and 18.04.

Jamey