cambridgehackers/connectal #185: PcieHost.bsv fails to build for multiple targets

RadhikaG, 2021-11-06:
### Tool versions
* bsc: [master](https://github.com/B-Lang-org/bsc)
* Connectal: master
* bsc-contrib: [master](https://github.com/B-Lang-org/bsc-contrib)

### Description

With `make build.vcu118`:
```
Error: "<path>/connectal/bsv/PcieHost.bsv", line 87, column 8: (T0031)
  The provisos for this expression could not be resolved because there are no
  instances of the form:
    Add#(0, 16, 32)
  The proviso was implied by expressions at the following positions:
    "<path>/connectal/bsv/PcieHost.bsv", line 88, column 26
Error: "<path>/connectal/bsv/PcieHost.bsv", line 127, column 4: (T0031)
  The provisos for this expression could not be resolved because there are no
  instances of the form:
    Connectable::Connectable#(ClientServer::Server#(PCIE::TLPData#(16), PCIE::TLPData#(16)), ClientServer::Client#(PCIE::TLPData#(32), PCIE::TLPData#(32)))
obj/Makefile:384: recipe for target 'obj/PcieHost.bo' failed
```

With `make build.vc707g2`:
```
Error: "<path>/connectal/bsv/PcieHost.bsv", line 87, column 8: (T0031)
  The provisos for this expression could not be resolved because there are no
  instances of the form:
    Add#(0, 16, 32)
  The proviso was implied by expressions at the following positions:
    "<path>/connectal/bsv/PcieHost.bsv", line 88, column 26
Error: "<path>/connectal/bsv/PcieHost.bsv", line 168, column 12: (T0020)
  Type error at:
    tlp

  Expected type:
    ClientServer::Client#(PCIE::TLPData#(ConnectalConfig::TlpDataBytes),
                      PCIE::TLPData#(ConnectalConfig::TlpDataBytes))

  Inferred type:
    ClientServer::Client#(PCIE::TLPData#(16), PCIE::TLPData#(16))

obj/Makefile:380: recipe for target 'obj/PcieHost.bo' failed
```

jameyhicks-cmt, 2021-11-08:
Please attach `git log -n 1` from `bsc` and `bsc-contrib`. I have not tested with open source `bsc` recently so I would like to know how to reproduce this.

jameyhicks-cmt, 2021-11-09:
I'm building from current head of `main`:
```
commit 91ad4276a74525410a9efe132785b812af5c9fc3 (HEAD -> main, origin/main)
Author: Julie Schwartz <quark@bluespec.com>
Date:   Sun Nov 7 17:24:54 2021 +1300

    Display the GHC version when building
```

jameyhicks-cmt, 2021-11-09:
Also, which version of vivado are you using?

jameyhicks-cmt, 2021-11-09:
I see the problem. Connectal depends on bsc-contrib being installed into the same place as bsc. I was able to reproduce this because I forgot to do that step.

Connectal also still depends on the BLUESPECDIR environment variable pointing to the "lib" directories which contains "Libraries".

jameyhicks-cmt, 2021-11-09:
To install bsc-contrib:
```
    make PREFIX=$PWD/../bsc/inst install

RadhikaG, 2021-11-09:
Hi Jamey, thanks for your reply!

So we've been trying to get open-source bluespec to play well with Connectal for an accelerator design we have (which previously used closed-source bluespec), for the vcu118 hardware target.

We grabbed the open source versions of bsc and bsc-contrib and installed them in the appropriate places, and got the open source stuff working in simulation (using the `build.verilator` target).

However we started hitting snags for `build.vcu118`. This is the first error we hit:
```
BSV_BO [<path>/connectal/generated/xilinx/PCIEWRAPPER3u.bsv]
make[1]: *** No rule to make target 'obj/XilinxVirtex7PCIE.bo', needed by 'obj/Pcie3EndpointX7.bo'.  Stop.
connectal/Makefile.connectal:57: recipe for target 'build.vcu118' failed
make: *** [build.vcu118] Error 2
```

I couldn't find XilinxVirtex7PCIE.bo in any of [B-Lang-Org](https://github.com/B-Lang-org/)'s official repos. I did however find it in an [ancient unofficial copy](https://github.com/GaloisInc/BESSPIN-BSC/blob/master/src/lib/BSVSource/Xilinx/XilinxVirtex7PCIE.bsv) of the bluespec source floating around on GitHub, and copied that into Connectal.

Compilation moved a bit further along but finally died at:
```
Error: "<path>/connectal/bsv/PcieHost.bsv", line 87, column 8: (T0031)
  The provisos for this expression could not be resolved because there are no
  instances of the form:
    Add#(0, 16, 32)
  The proviso was implied by expressions at the following positions:
    "<path>/connectal/bsv/PcieHost.bsv", line 88, column 26
Error: "<path>/connectal/bsv/PcieHost.bsv", line 127, column 4: (T0031)
  The provisos for this expression could not be resolved because there are no
  instances of the form:
    Connectable::Connectable#(ClientServer::Server#(PCIE::TLPData#(16), PCIE::TLPData#(16)), ClientServer::Client#(PCIE::TLPData#(32), PCIE::TLPData#(32)))
obj/Makefile:384: recipe for target 'obj/PcieHost.bo' failed
```

I decided to get the examples compiled to narrow down where exactly do things break. Most of the examples compiled (for the vcu118, vc707g2, and the vcu108 targets), except `memread256` and `memwrite256`, which died with the same error as above.

It turns out keeping `DataBusWidth=256` is the problem; our Connectal code uses `DataBusWidth=256` as well.

**The fix:**

We've temporarily downgraded our Connectal interface to `DataBusWidth=128` and everything compiles and runs okay for us.

However our Connectal interface did work with `DataBusWidth=256` with closed source bluespec, which leads me to suspect that either the open source version is missing some library, or Connectal itself isn't playing well with a bus size 256 wide.

It would be great if you could help us figure this out!

jameyhicks, 2021-11-09:
Hi Radhika,

Please file a separate but about DataBusWidth=256. I had this working a
long time ago, but the branch never got merged. I'm sure we can get it
working again.

Some of the Xilinx libraries were rejected from bsc-contrib for some
reason. I don't know why. Importing into Connectal seems like a way to deal
with it.

jameyhicks, 2021-11-09:
Also which version of Vivado are you using?

RadhikaG, 2021-11-09:
I'm using Vivado v.2020.1.

RadhikaG, 2021-11-10:
Also, I'll close this issue and open up another one for DataBusWidth=256 support. Would also like to figure out how to get the Xilinx libraries into open source bluespec (because those are pretty fundamental to getting the PCIe stuff working it seems!), but that I'll raise with the B-Lang-Org folks I guess.

closed 2021-11-10 by RadhikaG