cambridgehackers/connectal #186: DataBusWidth=256 support

RadhikaG, 2021-11-10:
For our accelerator design interfaced with Connectal, we've temporarily downgraded our interface to work with DataBusWidth=128 (it used to work with DataBusWidth=256 at some point with some ancient branch of Connectal...).

What's the best way to get DataBusWidth=256 working again? Is there a particular branch I should be looking at, or some patches I need to apply?

jameyhicks-cmt, 2021-11-10:
If you know which version of Connectal and Bluespec it was working with, that would help in finding a solution.

threonorm, 2021-11-10:
Hi @RadhikaG @jameyhicks-cmt ,

On the previous version of the design that @RadhikaG  is working on, I just remembered that I was actually not using upstream connectal, but Shuotao's fork:
https://github.com/xushuotao/connectal

(Sorry Radhika, I just remembered that yesterday, I had forgotten about it)

That version of connectal + some version of open source bluespec + vivado 2019.*, was allowing to compile the design with 256 width, I believe. But I am not sure it would still work.

RadhikaG, 2021-11-10:
ref xushuotao's fork: we were using that up till very recently; 256-width was not compiling with that either, it gave me the error below (the same one as issue #185):
```
Error: "<path>/connectal/bsv/PcieHost.bsv", line 87, column 8: (T0031)
  The provisos for this expression could not be resolved because there are no
  instances of the form:
    Add#(0, 16, 32)
  The proviso was implied by expressions at the following positions:
    "<path>/connectal/bsv/PcieHost.bsv", line 88, column 26
Error: "<path>/connectal/bsv/PcieHost.bsv", line 127, column 4: (T0031)
  The provisos for this expression could not be resolved because there are no
  instances of the form:
    Connectable::Connectable#(ClientServer::Server#(PCIE::TLPData#(16), PCIE::TLPData#(16)), ClientServer::Client#(PCIE::TLPData#(32), PCIE::TLPData#(32)))
obj/Makefile:384: recipe for target 'obj/PcieHost.bo' failed
```

I moved onto connectal master, since Shuotao's fork was giving us trouble with open source bluespec (which makes sense since the fork is relatively old) and I saw that current master supports the vcu118 as well.

jameyhicks, 2021-11-10:
I'll review Shuotao's branch and see if I can merge it in. I'll also look
at my databuswidth256 branch for relevant changes.

RadhikaG, 2021-11-10:
(to be clear, I personally never got Shuotao's branch working with closed source bluespec for DataBusWidth=256; I don't have access to closed source Bluespec either, I just know that @threonorm got it working at some point)

jameyhicks-cmt, 2021-11-10:
I think there are two sets of changes missing. Shuotao's branch has Ultrascale PCIE3 support and some level of DataBusWidth=256 support. The databuswidth256 branch enables full transfer rate.

jameyhicks, 2021-11-12:
I haven't looked at the file, but probably should be using https://github.com/GaloisInc/BESSPIN-BSC/blob/master/src/lib/BSVSource/Xilinx/XilinxVirtexUltraScalePCIE.bsv on UltraScale devices.