cambridgehackers/connectal #188: Error while running make command in connectal

manish-pp, 2022-08-19:
manish@manish:~/connectal$ make
(cd drivers/pcieportal; make)
make[1]: Entering directory '/home/manish/connectal/drivers/pcieportal'
make[2]: Entering directory '/usr/src/linux-headers-5.4.0-122-generic'
  CC [M]  /home/manish/connectal/drivers/pcieportal/portal.o
/home/manish/connectal/drivers/pcieportal/portal.c: In function ‘pcieportal_ioctl’:
/home/manish/connectal/drivers/pcieportal/portal.c:204:119: error: missing binary operator before token "("
 0)) && !(defined(RHEL_MAJOR) && RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0))
                                                                          ^
/home/manish/connectal/drivers/pcieportal/portal.c:210:119: error: missing binary operator before token "("
 0)) && !(defined(RHEL_MAJOR) && RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0))
                                                                          ^
scripts/Makefile.build:270: recipe for target '/home/manish/connectal/drivers/pcieportal/portal.o' failed
make[3]: *** [/home/manish/connectal/drivers/pcieportal/portal.o] Error 1
Makefile:1762: recipe for target '/home/manish/connectal/drivers/pcieportal' failed
make[2]: *** [/home/manish/connectal/drivers/pcieportal] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-122-generic'
Makefile:39: recipe for target 'pcieportal.ko' failed
make[1]: *** [pcieportal.ko] Error 2
make[1]: Leaving directory '/home/manish/connectal/drivers/pcieportal'
Makefile:32: recipe for target 'pciedrivers' failed
make: *** [pciedrivers] Error 2

I am getting the above error while running **make** command in connectal directory. Could anybody suggest some solution to this error?

jameyhicks, 2022-08-19:
Hmm, maybe that should have been defined(RHEL_MAJOR) instead of !defined(RHEL_MAJOR)
Or better yet, it should check for `defined(RHEL_RELEASE_CODE)`

jameyhicks, 2022-08-19:
I will fix the REDHAT ifdefs, but in the meantime you can build the driver from source using version v20.06.1. 

It would be helpful for me to know which FPGA you are using.

manish-pp, 2022-08-19:
"RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0)"

This part of the code is giving the error.

I am working with kintex-7 FPGA KC705

jameyhicks, 2022-08-19:
Connectal v20.06.1 should work well with the KC705. Later changes were for
VCU and AWS FPGAs.

manish-pp, 2022-08-19:
Thank you so much sir, I am now trying with Connectal v20.06.1.

closed 2022-08-20 by manish-pp