cambridgehackers/connectal #189: Error while building for kc705, "recipe for target 'obj/Pcie1EndpointX7.bo' failed"

manish-pp, 2022-08-21:
Error: "/home/manish/Desktop/Connectal/connectal-master/bsv/Pcie1EndpointX7.bsv", line 280, column 9: (G0007)
  Reference across clock domain in rule `txoutrule'.
  Method calls by clock domain:
    Clock domain 1:
      default_clock:
	c2b.o at "/home/manish/Desktop/Connectal/connectal-master/bsv/ConnectalClocks.bsv", line 77, column 16,
    Clock domain 2:
      b2c.c:
	b2c.inputclock at "/home/manish/Desktop/Connectal/connectal-master/bsv/ConnectalClocks.bsv", line 63, column 11,
  During elaboration of rule `txoutrule' at
  "/home/manish/Desktop/Connectal/connectal-master/bsv/Pcie1EndpointX7.bsv",
  line 280, column 9.
  During elaboration of `mkPcieEndpointX7' at
  "/home/manish/Desktop/Connectal/connectal-master/bsv/Pcie1EndpointX7.bsv",
  line 204, column 8.
obj/Makefile:586: recipe for target 'obj/Pcie1EndpointX7.bo' failed
make[1]: *** [obj/Pcie1EndpointX7.bo] Error 1
../../Makefile.connectal:57: recipe for target 'build.vc707' failed
make: *** [build.vc707] Error 2

I am getting the above error while building for kc705 board. I am using ubuntu 18.04.1 with latest connectal. Please help to resolve the error?

jameyhicks, 2022-08-22:
I got a different error when running `make build.kc705'.

Please try 'make build.kc705g2' which configures PCIe gen2 instead of gen1. That did build for me. I'll change the default for kc705 to gen2 because we stopped testing gen1 quite a while ago and I guess it has bit-rotted.

manish-pp, 2022-08-22:
Thank your sir, "make build.kc705g2" worked for my system. Now I have to test it on FPGA.

I am using the following system.

1. Ubuntu: Ubuntu 18.04.6 LTS
2. Kernel: 5.4.0-124-generic
3. Connectal: 22.05.23b, with "**&& RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(8,0)**" being commented out in files "portal.c", cdev_ctrl.c, xdma_mod.c, and portalmem.c
4. Vivado: Xilinx_Unified_2019.2_1106_2127_Lin64
5. Bluespec Compiler (build b024340)

closed 2022-08-22 by manish-pp

jameyhicks, 2022-08-22:
Hi Manish,

Please update to latest on master and rebuild the drivers before testing.

manish-pp, 2022-08-25:
> Hi Manish, Please update to latest on master and rebuild the drivers before testing.
> […](#)

Yes sir, I am using the updated one.