cambridgehackers/connectal #190: error while running "make run.kc705g2"

manish-pp, 2022-08-25:
manish@manish:~/Desktop/Connectal/connectal/examples/echo$ make run.kc705g2
grep: /home/manish/Desktop/Connectal/connectal/boardinfo/.json: No such file or directory
make -C kc705g2 --no-print-directory run
obj/Makefile:436: warning: overriding recipe for target 'verilog/mkDdr3.v'
obj/Makefile:135: warning: ignoring old recipe for target 'verilog/mkDdr3.v'
obj/Makefile:618: warning: overriding recipe for target 'verilog/mkPcieEndpointX7.v'
obj/Makefile:595: warning: ignoring old recipe for target 'verilog/mkPcieEndpointX7.v'
obj/Makefile:746: warning: overriding recipe for target 'verilog/mkPcieEndpointX7.v'
obj/Makefile:618: warning: ignoring old recipe for target 'verilog/mkPcieEndpointX7.v'
obj/Makefile:834: warning: overriding recipe for target 'verilog/mkPcieRootPortX7.v'
obj/Makefile:234: warning: ignoring old recipe for target 'verilog/mkPcieRootPortX7.v'
LD_PRELOAD=libSegFault.so SEGFAULT_USE_ALTSTACK=1 SEGFAULT_OUTPUT_NAME=bin/bsimexe-segv-output.txt /home/manish/Desktop/Connectal/connectal/scripts/run.pcietest ./bin/ubuntu.exe ; retcode=$?; exit $retcode
run.pcie parameters are: ./bin/ubuntu.exe
buffer /home/manish/Desktop/Connectal/connectal/examples/echo/kc705g2/bin/ubuntu.exe
ERROR: ld.so: object 'libSegFault.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
fpgajtag: elf input file, len 1078104 class 2
fpgajtag: unzip input file, len 603840
fpgajtag: Digilent:Digilent Adept USB Device:210203A7D954; bcd:700
count 0/1 cortex -1 dcount 0 trail 0
STATUS 00400019 done 0 release_done 0 eos 10 startup_state 0
fpgajtag: Starting to send file
fpgajtag: Done sending file
fpgajtag: bypass already programmed bc
STATUS 00401079 done 0 release_done 0 eos 10 startup_state 0
Running /usr/local/bin/pciescan.sh
+ PATH=/opt/tools/bsc/latest/bin:/home/manish/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/manish/Desktop/Connectal/android-ndk-r25/:/home/manish/shakti-tools/bin:/home/manish/shakti-tools/riscv32/bin:/home/manish/shakti-tools/riscv32/riscv32-unknown-elf/bin:/tools/Xilinx/Vivado/2019.2/bin:/sbin
+ for devid in 1be7:c100 1be7:b100
++ lspci -d 1be7:c100
++ sed -e 's/ .*//'
+ for devid in 1be7:c100 1be7:b100
++ lspci -d 1be7:b100
++ sed -e 's/ .*//'
+ sleep 1
+ lsmod
+ grep -q pcieportal
+ lsmod
+ grep -q bluenoc
+ sleep 1
+ sh -c 'echo 1 >/sys/bus/pci/rescan'
+ sleep 1
subprocess pid 10593 completed status=0 0
**Error: initPortalHardwareOnce: failed to open /dev/connectal, exiting
/home/manish/Desktop/Connectal/connectal/scripts/Makefile.connectal.build:250: recipe for target 'run' failed
make[1]: *** [run] Error 255
../../Makefile.connectal:68: recipe for target 'run.kc705g2' failed
make: *** [run.kc705g2] Error 2**

"**Make build.kc705g2**" worked fine but after connecting kintex board to my pc and running the command "**make run.kc705g2**", I am getting the above error. Please help me resolve the error....

jameyhicks-cmt, 2022-08-25:
One quirk of Linux is that it does not re-enumerate PCIe once it is booted, so you will have to reboot after the FPGA is programmed.

You only have to do this once. Subsequent reprogramming of the FPGA require a rescan but that is done automatically.

If you reprogrammed and rebooted, then run `lsmod | grep portal`. Both pcieportal and portalmem modules should be loaded.

jameyhicks-cmt, 2022-08-25:
Also, what is the output from l`spci -d 1be7:b100` ?

manish-pp, 2022-08-27:
Sir, I tried rebooting my PC after **Make run.kc705g2** command gave error, but still i am getting the same error message. I have one more doubt that, in which step, the FPGA is getting programmed? Is it getting programmed in **Make run.kc705g2** step?

I tried uploading bit-stream (kc705g2/Impl/TopDown/mkTop.bit) using vivado, then i rebooted but still getting same error.

"lsmod | grep portal" and "lspci -d 1be7:b100" are not giving any output.

jameyhicks, 2022-08-27:
Please try `lspci` to list all the PCIe devices.

manish-pp, 2022-08-29:
> Please try `lspci` to list all the PCIe devices.

The kintex board is not showing in the list, but when I am doing **lsusb**, It is showing.

jameyhicks-cmt, 2022-08-29:
OK, then I would recheck the connections:

- reseat kc705 in pcie slot or reconnect pcie cable if using one
- make sure kc705 stays powered when PC reboots

jameyhicks, 2022-08-29:
To answer an earlier question, the FPGA gets programmed when you run make run.kc705g2. However, I am glad that you tried manual programming to verify that it was programmed before retrying the other steps.

manish-pp, 2022-08-30:
> OK, then I would recheck the connections:
> 
>     * reseat kc705 in pcie slot or reconnect pcie cable if using one
> 
>     * make sure kc705 stays powered when PC reboots

manish@manish:~/Desktop/Connectal/connectal/examples/echo$ make run.kc705g2 V=1
grep: /home/manish/Desktop/Connectal/connectal/boardinfo/.json: No such file or directory
make -C kc705g2 --no-print-directory run
obj/Makefile:436: warning: overriding recipe for target 'verilog/mkDdr3.v'
obj/Makefile:135: warning: ignoring old recipe for target 'verilog/mkDdr3.v'
obj/Makefile:618: warning: overriding recipe for target 'verilog/mkPcieEndpointX7.v'
obj/Makefile:595: warning: ignoring old recipe for target 'verilog/mkPcieEndpointX7.v'
obj/Makefile:746: warning: overriding recipe for target 'verilog/mkPcieEndpointX7.v'
obj/Makefile:618: warning: ignoring old recipe for target 'verilog/mkPcieEndpointX7.v'
obj/Makefile:834: warning: overriding recipe for target 'verilog/mkPcieRootPortX7.v'
obj/Makefile:234: warning: ignoring old recipe for target 'verilog/mkPcieRootPortX7.v'
LD_PRELOAD=libSegFault.so SEGFAULT_USE_ALTSTACK=1 SEGFAULT_OUTPUT_NAME=bin/bsimexe-segv-output.txt /home/manish/Desktop/Connectal/connectal/scripts/run.pcietest ./bin/ubuntu.exe ; retcode=$?; exit $retcode
run.pcie parameters are: ./bin/ubuntu.exe
buffer /home/manish/Desktop/Connectal/connectal/examples/echo/kc705g2/bin/ubuntu.exe
**ERROR: ld.so: object 'libSegFault.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.**
fpgajtag: elf input file, len 1078104 class 2
fpgajtag: unzip input file, len 603840
fpgajtag: Digilent:Digilent Adept USB Device:210203A7D954; bcd:700
count 0/1 cortex -1 dcount 0 trail 0
STATUS 00401079 done 0 release_done 0 eos 10 startup_state 0
fpgajtag: Starting to send file
fpgajtag: Done sending file
fpgajtag: bypass already programmed bc
STATUS 00401079 done 0 release_done 0 eos 10 startup_state 0
Running /usr/local/bin/pciescan.sh
\+ PATH=/opt/tools/bsc/latest/bin:/home/manish/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/manish/Desktop/Connectal/android-ndk-r25/:/home/manish/shakti-tools/bin:/home/manish/shakti-tools/riscv32/bin:/home/manish/shakti-tools/riscv32/riscv32-unknown-elf/bin:/tools/Xilinx/Vivado/2019.2/bin:/sbin
\+ for devid in 1be7:c100 1be7:b100
\++ lspci -d 1be7:c100
\++ sed -e 's/ .*//'
\+ for devid in 1be7:c100 1be7:b100
\++ lspci -d 1be7:b100
\++ sed -e 's/ .*//'
\+ sleep 1
\+ lsmod
\+ grep -q pcieportal
\+ lsmod
\+ grep -q bluenoc
\+ sleep 1
\+ sh -c 'echo 1 >/sys/bus/pci/rescan'
\+ sleep 1
subprocess pid 9301 completed status=0 0
**Error: initPortalHardwareOnce: failed to open /dev/connectal, exiting
/home/manish/Desktop/Connectal/connectal/scripts/Makefile.connectal.build:250: recipe for target 'run' failed
make[1]: *** [run] Error 255
../../Makefile.connectal:68: recipe for target 'run.kc705g2' failed
make: *** [run.kc705g2] Error 2**

I tried restarting PC many times but still getting the same above error messages, below is the output of "lsmod | grep portal" and there is no output for "lspci -d 1be7:b100"

manish@manish:~/Desktop/Connectal/connectal/examples/echo$ lsmod | grep portal
portalmem              16384  0

manish-pp, 2022-08-30:
> To answer an earlier question, the FPGA gets programmed when you run make run.kc705g2. However, I am glad that you tried manual programming to verify that it was programmed before retrying the other steps.

So the FPGA is getting detected and programmed in the following steps while running "**make run.kc705g2**"?

fpgajtag: elf input file, len 1078104 class 2
fpgajtag: unzip input file, len 603840
fpgajtag: Digilent:Digilent Adept USB Device:210203A7D954; bcd:700
count 0/1 cortex -1 dcount 0 trail 0
STATUS 00401079 done 0 release_done 0 eos 10 startup_state 0
fpgajtag: Starting to send file
fpgajtag: Done sending file
fpgajtag: bypass already programmed bc