cambridgehackers/connectal #191: ERROR: could not insert 'portalmem'

manish-pp, 2022-08-29:
manish@manish:~/Desktop/Connectal/connectal$ sudo modprobe portalmem
[sudo] password for manish: 
modprobe: ERROR: could not insert 'portalmem': Operation not permitted

In the step of installing drivers (**Preparation for Kintex and Virtex boards**), "**sudo modprobe portalmem**" command is giving the above error. Please help me resolve the error.

jameyhicks, 2022-08-29:
I think if you run `dmesg` you will find that your kernel has updated since you built portalmem.ko. If so, rebuild the module and you should be able to insmod / modprobe the module.

manish-pp, 2022-08-30:
> I think if you run `dmesg` you will find that your kernel has updated since you built portalmem.ko. If so, rebuild the module and you should be able to insmod / modprobe the module.

I disabled the "Secure Boot" in BIOS setting, now I am not getting any error but there is not output also.

closed 2022-08-30 by manish-pp