cambridgehackers/connectal #196: Problem while building for zedboard

Arch1610, 2023-04-16:
While building the code for zedboard, I am facing the following error:

File "/home/archana/connectal-latest/scripts/reorderbytes.py", line 36, in <module>
    readarr = array.array('I', infile.read())
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
/home/archana/connectal-latest/scripts/Makefile.connectal.build:201: recipe for target 'bits' failed
make[1]: *** [bits] Error 1
../../Makefile.connectal:57: recipe for target 'build.zedboard' failed
make: *** [build.zedboard] Error 2

Could anybody pls suggest a possible solution for this issue

[image]

jankcorn, 2023-04-16:
the default file mode in python was changed to utf8, so the script reorderbytes.py has been updated to force input and output files to be binary.
Please pull the update and let me know if this works for you as well.

Arch1610, 2023-04-17:
Thank you so much.
It worked.

closed 2023-04-17 by jankcorn