cambridgehackers/connectal #197: obsolete bsc version checking bug in `scripts/Makefile.connectal.build`

01xz, 2023-10-15:
hi,

i'm now using the latest open-source Bluespec Compiler, `bsc -v` returns
```
Bluespec Compiler, version 2023.07 (build 0eb551d)
This is free software; for source code and copying conditions, see
https://github.com/B-Lang-org/bsc

Invoking command line:
bsc -v
```

however, the condition in [L76-L78](https://github.com/cambridgehackers/connectal/blob/66d30ffcca8e0260041fb6b48b3200411ff73282/scripts/Makefile.connectal.build#L76-L78) in `scripts/Makefile.connectal.build` will be met,
and will think of bsc version 2023.07 as obsolete,
due to

`BSCVERSION` in [L72](https://github.com/cambridgehackers/connectal/blob/66d30ffcca8e0260041fb6b48b3200411ff73282/scripts/Makefile.connectal.build#L72) is `2023.07`
`BSCMAJOR` in [L73](https://github.com/cambridgehackers/connectal/blob/66d30ffcca8e0260041fb6b48b3200411ff73282/scripts/Makefile.connectal.build#L73) is `2023`

and `$(shell test $(BSCMAJOR) -le 2019 &>/dev/null && echo obsolete)` will return empty and the `ifeq` will be met in [L76](https://github.com/cambridgehackers/connectal/blob/66d30ffcca8e0260041fb6b48b3200411ff73282/scripts/Makefile.connectal.build#L76)

i think it's a bug here, caused by [#192](https://github.com/cambridgehackers/connectal/pull/192)

jameyhicks, 2023-10-16:
Good catch!

jameyhicks, 2023-10-16:
Fixed

closed 2023-10-16 by jameyhicks